pschatzmann/arduino-snapclient #7: Stuttery output

h3ndrik, 2024-02-24:
I followed the example code closely and set up an ESP32 with PSRAM and an external I2S DAC. Project is built in PlatformIO and I've made sure the CPU is set to 240MHz.

Sound output is very stuttery, from unusable to unbearable. What am I doing wrong?
* I've tried setting the snapserver buffer length to 1000 and 500
* I've tried 3 codecs. The best result is with Opus, followed by FLAC and worst is PCM

pschatzmann, 2024-02-24:
I think you are pulling your own leg by having the log level set to info. This is impacting the audio quality.
Make sure that it is set to warning or error

h3ndrik, 2024-02-25:
Yeah, I just set the loglevel to INFO to be able to provide logs. I've reverted that and set it to Error and it doesn't change anything. Issue remains the same.

pschatzmann, 2024-02-25:
Strange! Can it be that you have a bad wifi connection that is slowing things down ?
Did you try to repeat my testing steps ? 

The current implementation is not using any buffering. The server sends the audio data which to the decoder which just forwards the result to the i2s output. So the only place where you could increase the buffer size is in the I2S configuration.

Of cause it would be pretty easy to subclass the SnapOutput class and extend it with a buffer, but I guess this would not really help if the sending from the server is too slow. 

You could also try to play with the different Synchonization implemntations:  e.g static SnapTimeSyncFixed synch(123, 0.9);
client.begin(synch); is slowing down the playback by 10%

h3ndrik, 2024-02-26:
Thank you. Somehow I missed that Wiki page about testing. I've done that in the meantime. I'm using 44100:16:2 now.

I had hoped you'd provide me with something that newbies to the arduino-audio-lib frequently get wrong... Maybe I have to wrap my head around how all the internals tie together and add a few debug printf statements to get to the root.

I've restarted my wifi router, I've moved the ESP32 closer to the router and installed another Snapserver on my laptop and configured it like on the Wiki page. I've tried SnapTimeSyncFixed and PSRAM and no Psram, I've fiddled with many values but I can only make it worse.

I'm using an [Lolin D32 Pro](https://www.wemos.cc/en/latest/d32/d32_pro.html) and a MAX98357A I2S amp, which is described in the arduino-audio-tools wiki.

I didn't understand that SnapClient::doLoop() was a blocking method. I've moved that into another task with a decent stack size, and I'm trying to use the main loop to get some debug information every few seconds... What I found out already:

* ESP.getFreeHEAP() reports 196160
* ESP.getFreePsram() reports 4137623
* i2s.available() => 3072
* i2s.availableForWrite => 3072

The last two can't be right?! Doesn't that mean the i2s dma buffer is next to empty and underrunning? it is currently configured to be 6*512. But if it were empty all the time, it'd explain why changing the buffers doesn't do anything. What's the next step, see what the Decoder does?

pschatzmann, 2024-02-26:
Can you attach some recorded example how it sounds, so that I can get a better feeling.
The i2s.available() and availableForWrite() just return a fixed value because the ESP has no API to provide this information and I am using a blocking read and blocking write so that this amount is always working.

h3ndrik, 2024-02-26:
Ah, thx. Yeah I'm currently wrapping up things for today. I'll see if I can get some more info tomorrow. I also fiddled with the APLL and logging and currently everything is a mess ;-)

pschatzmann, 2024-02-26:
I committed some additional DRAFT code that implements a processor which uses FreeRTOS to buffer the encoded data.
If you can't resolve your issue by increasing the I2S buffer parameters, then this might be another option.

h3ndrik, 2024-02-28:
Thank you. Sorry for this lengthy discussion. I've attached logs below... But I think I should take a step back and replicate your proven setup before wasting more of our time... I think I'm currently doing too many things at once. I'll clean up my development environment first.

May I have the details of the setup you use to compile this?

* Version of the Arduino core you'd like me to target? (I'm not sure what PlatformIO deems appropriate for me...)
* What IDE / framework do you use to develop this? Do you use the Arduino IDE? Which one is it, Legacy 1.8.x or 2.x ?

---

 I didn't have any success with the FreeRTOS code, yet:

```
ELF file SHA256: 97c69dc2502e6733

E (416) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

assert failed: xStreamBufferGenericCreate stream_buffer.c:299 (xBufferSizeBytes > 0)

Backtrace: 0x40083b71:0x3ffe3a50 0x4008f09d:0x3ffe3a70 0x400951dd:0x3ffe3a90 0x4009268d:0x3ffe3bc0 0x400dd4c9:0x3ffe3be0 0x400dd76c:0x3ffe3c00 0x400febcb:0x3ffe3c20 0x40083285:0x3ffe3c50 0x40079316:0x3ffe3c90 |<-CORRUPTED

  #0  0x40083b71:0x3ffe3a50 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008f09d:0x3ffe3a70 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x400951dd:0x3ffe3a90 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x4009268d:0x3ffe3bc0 in xStreamBufferGenericCreate at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/stream_buffer.c:299 (discriminator 1)
  #4  0x400dd4c9:0x3ffe3be0 in audio_tools::SynchronizedBufferRTOS<unsigned char>::SynchronizedBufferRTOS(unsigned int, unsigned int, unsigned int, unsigned int) at .pio/libdeps/lolin_d32_pro/- arduino-audio-tools/src/AudioTools/SynchronizedBuffers.h:348
      (inlined by) SnapProcessorRTOS::SnapProcessorRTOS(int, int) at .pio/libdeps/lolin_d32_pro/- arduino-snapclient/src/api/SnapProcessorRTOS.h:23
      (inlined by) __static_initialization_and_destruction_0 at src/main.cpp:20
  #5  0x400dd76c:0x3ffe3c00 in _GLOBAL__sub_I__ZN17SnapProcessorRTOS4selfE at src/main.cpp:99
  #6  0x400febcb:0x3ffe3c20 in do_global_ctors at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:201
      (inlined by) start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:444
  #7  0x40083285:0x3ffe3c50 in call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/cpu_start.c:679
  #8  0x40079316:0x3ffe3c90 in ?? ??:0
```

And in the normal example I had to add `client.setServerIP()` now. MDNS or setting CONFIG_SNAPCAST_SERVER_HOST seemed not to be enough anymore?!

```
[  3655][I][SnapClient.h:98] begin(): [SnapClient] Connected to AP
[  3656][I][SnapClient.h:222] setupMACAddress(): [SnapClient] mac: B4:E6:2D:A4:6D:E9
[  6765][I][SnapClient.h:183] setupMDNS(): [SnapClient] SNAPCAST ip: 172.16.33.20, port: 1704
[  7839][I][SnapTime.h:62] printLocalTime(): [SnapTime] SNTP: Time is 28-02-2024 18-22-18
[ 10842][I][WiFiClient.cpp:260] connect(): select returned due to timeout 3000 ms for fd 48
[ 10842][E][SnapProcessor.h:217] connectClient(): [SnapProcessor] ... socket connect failed errno=119
[ 17861][I][WiFiClient.cpp:260] connect(): select returned due to timeout 3000 ms for fd 48
[ 17861][E][SnapProcessor.h:217] connectClient(): [SnapProcessor] ... socket connect failed errno=119
[ 24880][I][WiFiClient.cpp:260] connect(): select returned due to timeout 3000 ms for fd 48
[ 24880][E][SnapProcessor.h:217] connectClient(): [SnapProcessor] ... socket connect failed errno=119
```

And if I do, it also boot-loops now:

```
ELF file SHA256: da66da10cd850e9c

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
[... 39 lines cut ...]
PC      : 0x400d7507  PS      : 0x00060530  A0      : 0x800d9bc5  A1      : 0x3ffb3190  
A2      : 0x3ffc3d38  A3      : 0x000001f4  A4      : 0x00000000  A5      : 0x3f4001ce  
A6      : 0x00000000  A7      : 0x00010000  A8      : 0x800d74ff  A9      : 0x3ffb3170  
A10     : 0x0000199c  A11     : 0x0000199c  A12     : 0x3f4001ce  A13     : 0x0000002e  
A14     : 0x3f401932  A15     : 0x3f40157e  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x40085bc1  LEND    : 0x40085bc9  LCOUNT  : 0x00000027  

Backtrace: 0x400d7504:0x3ffb3190 0x400d9bc2:0x3ffb31e0 0x400d9d18:0x3ffb3200 0x400d9d2d:0x3ffb3250 0x400d69c7:0x3ffb3270 0x400ecc89:0x3ffb3290

  #0  0x400d7504:0x3ffb3190 in SnapTimeSync::setMessageBufferDelay(int) at .pio/libdeps/lolin_d32_pro/- arduino-snapclient/src/api/SnapTimeSync.h:46 (discriminator 6)
      (inlined by) SnapProcessor::processMessageServerSettings() at .pio/libdeps/lolin_d32_pro/- arduino-snapclient/src/api/SnapProcessor.h:435 (discriminator 6)
  #1  0x400d9bc2:0x3ffb31e0 in SnapProcessor::processMessageLoop() at .pio/libdeps/lolin_d32_pro/- arduino-snapclient/src/api/SnapProcessor.h:190
[... cut ...]
  #5  0x400ecc89:0x3ffb3290 in loopTask(void*) at /home/hendrik/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

Unrelated to that, I found the following in the SnapServer log, regarding the sample rates, after I've changed to another Snapserver version on another (older Debian) computer:
```
Feb 28 18:15:43 host snapserver[20350]: Resampling input from 44100:16:2 to 48000:16:2 as required by Opus
Feb 28 18:15:43 host snapserver[20350]: Resampling from 44100:16:2 to 48000:16:2
Feb 28 18:15:43 host snapserver[20350]: Opus bitrate: 192000 bps, complexity: 10
```

pschatzmann, 2024-02-28:
The error message: assert failed: xStreamBufferGenericCreate stream_buffer.c:299 (xBufferSizeBytes > 0) comes from a missing update of the AudioTools. I only recently added the functionality to support the creation of an empty buffer which is resized later.

I tested in Arduino with the ESP32 version 2.0.11 and 2.0.14. 
The Arduino IDE version does not matter.

pschatzmann, 2024-02-28:
You are right: the server IP logic was broken because I needed to change the data flow to support the new processor class.
The correction has been committed

h3ndrik, 2024-02-28:
Wow, I've never had that many annoyances with PlatformIO. Thank you.

Additionally, it turns out PlatformIO doesn't choose a sane version of the Arduino framework per default. And my previous attempt at changing that somehow failed. I managed to set the version of the Arduino framework now and it sounds way better!

If I choose `SnapTimeSyncFixed synch(CONFIG_PROCESSING_TIME_MS, 1.0);` it plays almost fine but constantly makes some minor crackling noises. I think that's some resampling or dropping small chunks? It doesn't seem to apply the 1.0 correctly.
But when I use `SnapTimeSyncDynamic synch(172, 10);`, it'll intermittendly say the following and the sound becomes crystal clear:

```
[ 23496][I][SnapTimeSync.h:113] getFactor(): [SnapTimeSync] => adjusting playback speed by factor: 1.000000
[I] ResampleStream.h : 267 - setStepSize: 1.000000
```

So I think I almost got it. My main issue seems to have been the Arduino framework version.

Two more things about the recent changes:
* The 'classic' way stops bootlooping only if I explicitly pass `client.begin()` a SnapTimeSync object. Maybe that's a regression?
* And the manual setting of `client.setServerIP()` also doesn't seem right.

pschatzmann, 2024-02-28:
I also noticed some rare hickups and I think this might come when the i2s buffer is running empty, so increasing it might help.
With the RTOS buffer version however there is no such issue any more...

h3ndrik, 2024-02-28:
Does the PROCESSING_TIME_MS work with the SnapProcessorRTOS? The ESP is always ahead and it doesn't seem to make a difference what I set there.

pschatzmann, 2024-02-29:
It should, but because I am also always confused how this is working, I reversed the logic today:
- now increasing the value is adding some additional delay and 
- decreasing the value is removing the delay (negative values are possible) 

Before it was just the other way around.

If you use the SnapProcessorRTOS you can also play with the buffer size to adjust the delay

h3ndrik, 2024-02-29:
I found another thing that seems related to my remaining crackling... Why is that a funny number and not `1.0f`?

https://github.com/pschatzmann/arduino-snapclient/blob/0a8df39a5fe37a84aadf36fe52fccc2f8f6c94ae/src/api/SnapOutput.h#L169

Additionally: SnapTimeSyncFixed seems to be constantly resampling with that... Is it correct that SnapTimeSync::isSync() can never be true with the SyncFixed? Since it never increases update_count contrary to the other classes...

pschatzmann, 2024-02-29:
You are right, it should be 1.0f. I guess I was playing around and forget to reset the value...

pschatzmann, 2024-02-29:
It is supposed to be     res_cfg.step_size = p_snap_time_sync->getFactor();

h3ndrik, 2024-03-01:
Thanks a lot for all the help! I'm closing this now. If someone else reads this: If you're using PlatformIO, make sure you're using a recent Arduino core. That was my very first issue. And use Opus. I've included the most important settings below, I'll try to contribute a more complete PlatformIO example once I get it working.

```ini
[env:lolin_d32_pro]
platform = https://github.com/platformio/platform-espressif32.git#develop
board = lolin_d32_pro
framework = arduino
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.14

lib_deps =
    arduino-snapclient=https://github.com/pschatzmann/arduino-snapclient
    arduino-audio-tools=https://github.com/pschatzmann/arduino-audio-tools
    arduino-libopus=https://github.com/pschatzmann/arduino-libopus

build_flags =
    -D USE_AUDIO_LOGGING=true
    -D LOG_LEVEL="AudioLogger::Warning"
    -D LOG_STREAM="Serial"
    -D LOG_LOCAL_LEVEL=ESP_LOG_WARNING
    -D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARNING
    -D CONFIG_ARDUHAL_LOG_COLORS
```

closed 2024-03-01 by h3ndrik

pschatzmann, 2024-03-01:
Did you use SnapProcessorRTOS or SnapProcessor ?

h3ndrik, 2024-03-02:
I've switched to SnapProcessorRTOS not long after we talked about that. And I'm using Opus with its single supported bitrate of 48000. That seems to be the only working codec as of now. I've tested both a Snapserver in the local network and an unusual setup where it's running on a VPS on the internet. Both seem to work.