pschatzmann/arduino-snapclient #14: 'SnapClientI2S' example crashes

constant-flow, 2024-03-13:
Hi thanks for this project, as I'd love to replace my Snapcast RaspberryPis with lower power ESP32s

I can't get your code to run without crashes and I don't know how to debug further.
I basically took your code into a platformio project to get a better understanding where it's crashing.

this is what i get

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Connecting to WiFi ...
192.168.178.75
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d6cb4  PS      : 0x00060f30  A0      : 0x800d91e2  A1      : 0x3ffb2200  
A2      : 0x3ffc3c98  A3      : 0x000003e8  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffc4930  A8      : 0x800d6c9f  A9      : 0x3ffb21c0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x0000001e  A13     : 0x3ffc3cf8  
A14     : 0x3ffc3d68  A15     : 0x00010000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000001c  LBEG    : 0x4008afd9  LEND    : 0x4008afe9  LCOUNT  : 0xffffffff  

Backtrace: 0x400d6cb1:0x3ffb2200 0x400d91df:0x3ffb2220 0x400d5e8d:0x3ffb2270 0x400eac69:0x3ffb2290

  #0  0x400d6cb1:0x3ffb2200 in snap_arduino::SnapProcessor::processMessageServerSettings() at src/api/SnapProcessor.h:445
      (inlined by) snap_arduino::SnapProcessor::processMessageServerSettings() at src/api/SnapProcessor.h:425
  #1  0x400d91df:0x3ffb2220 in snap_arduino::SnapProcessor::processMessageLoop() at src/api/SnapProcessor.h:201
      (inlined by) snap_arduino::SnapProcessor::processMessageLoop() at src/api/SnapProcessor.h:173
      (inlined by) snap_arduino::SnapProcessor::processLoopStep() at src/api/SnapProcessor.h:145
      (inlined by) snap_arduino::SnapProcessor::processLoopStep() at src/api/SnapProcessor.h:128
      (inlined by) snap_arduino::SnapProcessor::doLoop() at src/api/SnapProcessor.h:73
  #2  0x400d5e8d:0x3ffb2270 in snap_arduino::SnapClient::doLoop() at src/SnapClient.h:154
      (inlined by) loop() at src/main.cpp:57
  #3  0x400eac69:0x3ffb2290 in loopTask(void*) at /home/bb8/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

Which is this line `SnapProcessor.h:445`
```
p_snap_output->snapTimeSync().setMessageBufferDelay(
        server_settings_message.buffer_ms + server_settings_message.latency);
```

I see the Snapcast client pop up quickly in the server's UI, then it crashes and disappears.
Any ideas how to get around that error? I tried with a `WROOM-32` and a `WROVER-B` board

Could it be a completely different section in which this error originates or a memory issue? Because I tried to pull that line into three lines and then the sum alone gave me that exception. Thanks in advance.

pschatzmann, 2024-03-13:
Try to use the latest committed version. In PlatformIO you might need to delete the folder with the library to trigger a reload. Alternatively just give a synchronizer as parameter when calling begin. 

Or better try to use the RTOS example

constant-flow, 2024-03-13:
I installed basically everything new this week, so there shouldn't be a newer version i missed. So for me the Non-RTOS version seems to be not working.

Good news though, RTOS works, but not in sync with the other speakers yet. They should be in synced, right?

constant-flow, 2024-03-13:
Just found that wiki entry and will also check if the non-rtos version runs as expected.
Your are damn fast :star_struck: thanks for such a support

constant-flow, 2024-03-13:
Seems to be fixed with this code changes :smile: 

Remaining issue is more about synchronizing and constant throughput, so this can be closed.
(Props to you, never had such an instant support via github)

closed 2024-03-13 by constant-flow