pschatzmann/arduino-snapclient #24: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled. when using ETH.h

vladniko, 2024-07-03:
I encounter an error while trying to use ESP32 with ETHERNET

`ETH Started
ETH Connected

PCM51xx initialized successfully.
Setup Exited. Next hop: loop()
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d69c9  PS      : 0x00060230  A0      : 0x800d9edb  A1      : 0x3ffb21b0  
A2      : 0x3ffc3b90  A3      : 0x9e3a431e  A4      : 0x00000010  A5      : 0x3ffc4db0  
A6      : 0x00000000  A7      : 0x330a000a  A8      : 0x800e151c  A9      : 0x3ffb20f0  
A10     : 0x00000000  A11     : 0x3f408ac0  A12     : 0x0000001e  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008b4e5  LEND    : 0x4008b4f5  LCOUNT  : 0xffffffff  

Backtrace: 0x400d69c6:0x3ffb21b0 0x400d9ed8:0x3ffb2230 0x400d6d7d:0x3ffb2250 0x400e19f4:0x3ffb2270 0x40091676:0x3ffb2290

ELF file SHA256: 9858541605b79ab7

Rebooting...
`

My source code is:

`#include "AudioTools.h"
#include "SnapClient.h"
//#include "AudioCodecs/CodecOpus.h" // https://github.com/pschatzmann/arduino-libopus
#include "api/SnapProcessorRTOS.h" // install https://github.com/pschatzmann/arduino-freertos-addons

#include "PCM51xx.h"
#include "Wire.h"

#ifndef ETH_PHY_TYPE
#define ETH_PHY_TYPE  ETH_PHY_LAN8720
#define ETH_PHY_ADDR  0
#define ETH_PHY_MDC   23
#define ETH_PHY_MDIO  18
#define ETH_PHY_POWER -1
#define ETH_CLK_MODE  ETH_CLOCK_GPIO17_OUT
#endif

#include <ETH.h>

WAVDecoder codec;

//WiFiClient wifi;

static bool eth_connected = false;
NetworkClient ethclient;

I2SStream out;

SnapProcessorRTOS rtos(1024*8); // define queue with 8 kbytes
SnapClient client(ethclient, out, codec);
PCM51xx pcm(Wire);

void setup() {
  Serial.begin(115200);
  
  // login to wifi -> Define values in SnapConfig.h or replace them here
  //WiFi.begin("wifi", "pass");
//  Serial.print("Connecting to WiFi ..");
//  while (WiFi.status() != WL_CONNECTED) {
//    Serial.print('.');
//    delay(1000);
//  }

  Network.onEvent(onEvent);
  ETH.begin();
  
  // print ip address
  Serial.println();
  //Serial.println(WiFi.localIP());

  // setup I2S to define custom pins
  auto cfg = out.defaultConfig();
  cfg.sample_rate = 44100;
  cfg.bits_per_sample = 16;
  cfg.channels = 2;
  cfg.pin_bck = 2;
  cfg.pin_ws = 5;
  cfg.pin_data = 14;
  cfg.buffer_size = 1536;
  cfg.buffer_count = 60;
  out.begin(cfg);

  Wire.begin(13, 16); //HW dependent ! SDA, SCL

  if (pcm.begin(PCM51xx::SAMPLE_RATE_48K, PCM51xx::BITS_PER_SAMPLE_16))
    Serial.println("PCM51xx initialized successfully.");
  else
  {
    pcm.setPowerMode(PCM51xx::POWER_MODE_ACTIVE);
    Serial.println("Failed to initialize PCM51xx.");
    uint8_t powerState = pcm.getPowerState();
    if (powerState == PCM51xx::POWER_STATE_I2C_FAILURE)
    {
      Serial.print("No answer on I2C bus at address ");
      Serial.println(pcm.getI2CAddr());
    }
    else
    {
      Serial.print("Power state : ");
      Serial.println(pcm.getPowerState());
      Serial.println("Check that the sample rate / bit depth combination is supported.");
    }
  }

  //Set volume
  pcm.setVolume(48);

  // Use FreeRTOS
  client.setSnapProcessor(rtos);

  // Define CONFIG_SNAPCAST_SERVER_HOST in SnapConfig.h or here
  client.setServerIP(IPAddress(10,0,10,50));

  // start snap client
  client.begin();
  Serial.println("Setup Exited. Next hop: loop()");
  
}

void loop() {
  client.doLoop();
}

// WARNING: onEvent is called from a separate FreeRTOS task (thread)!
void onEvent(arduino_event_id_t event) {
  switch (event) {
    case ARDUINO_EVENT_ETH_START:
      Serial.println("ETH Started");
      // The hostname must be set after the interface is started, but needs
      // to be set before DHCP, so set it from the event handler thread.
      ETH.setHostname("esp32-ethernet");
      break;
    case ARDUINO_EVENT_ETH_CONNECTED: Serial.println("ETH Connected"); break;
    case ARDUINO_EVENT_ETH_GOT_IP:
      Serial.println("ETH Got IP");
      Serial.println(ETH);
      eth_connected = true;
      break;
    case ARDUINO_EVENT_ETH_LOST_IP:
      Serial.println("ETH Lost IP");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default: break;
  }
}`

pschatzmann, 2024-07-03:
You are the only one who can decode the backtrace !

pschatzmann, 2024-07-19:
https://github.com/pschatzmann/arduino-snapclient/issues/26#issuecomment-2238069285

closed 2024-07-19 by pschatzmann