pschatzmann/arduino-snapclient #25: ethernet bootloop

parabyte, 2024-07-17:
im using the espressive built in ETH.H library, i believe i have done things correctish all thats happening is a bootloop on my esp32

/**
 * @brief SnapClient with pcm decoder: I2S Output on an ESP32
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "SnapClient.h"
#define ARDUINO_LOOP_STACK_SIZE (10 * 1024)
#include <SPI.h>

#include <ETH.h>

#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS   5
#define ETH_PHY_IRQ  22
#define ETH_PHY_RST  21

// SPI pins
#define ETH_SPI_SCK  18
#define ETH_SPI_MISO 23
#define ETH_SPI_MOSI 19

 WiFiClient ethernet;

WAVDecoder pcm;
I2SStream out;
SnapTimeSyncDynamic synch(172, 10); // optional configuration
SnapClient client(ethernet, out, pcm);

void ethernetWizReset(const uint8_t resetPin) {
    pinMode(resetPin, OUTPUT);
    digitalWrite(resetPin, HIGH);
    delay(250);
    digitalWrite(resetPin, LOW);
    delay(50);
    digitalWrite(resetPin, HIGH);
    delay(350);
}

void macCharArrayToBytes(const char* str, byte* bytes) {
    for (int i = 0; i < 6; i++) {
        bytes[i] = strtoul(str, NULL, 16);
        str = strchr(str, ':');
        if (str == NULL || *str == '\0') {
            break;
        }
        str++;
    }
}

void connectEthernet() {
        ETH.setHostname("esp32-eth0");

  //  ethernetWizReset(RESET);
  
    Serial.println("Starting ETHERNET connection...");
  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);

    Serial.print("Ethernet IP is: ");
    Serial.println(ethernet.localIP());
}

void setup() {
    Serial.begin(115200);

  SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
    connectEthernet();

    auto cfg = out.defaultConfig(TX_MODE);
    cfg.pin_bck = 21;
    cfg.pin_ws = 19;
    cfg.pin_data = 26;
    cfg.sample_rate = 44100; // Set the sample rate to 44.1 kHz
    cfg.bits_per_sample = 16;  // Set bits per sample to 16
    cfg.i2s_format = I2S_PHILIPS_FORMAT;  // Stereo channel, Philips format (I2S)
    cfg.use_apll = true;  // Use the audio PLL for low-jitter
    cfg.is_master = true;  // Set to master mode
    out.begin(cfg);

    client.begin(synch);
}

void loop() {
    client.doLoop();
}

pschatzmann, 2024-07-17:
I never used this, but I would have the following suggestions:
- check the result of the ETH.begin call
- I think the client is wrong: The [esp32 example](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino) uses NetworkClient: so replace the WiFiClient!
- If you use the WavDecoder, it is not necessary to increase the stacksize

closed 2024-07-17 by pschatzmann

parabyte, 2024-07-17:
okay the above code wont fit into my esp32, so i given this code a try with the suggestions you made! 

the below code fits and i have enough space to spare to turn on debugging but unfortunately not enough space for verbose debugging messages.
[image]

I am getting a ip via dhcp. with a hostname but then the code crashes out.

#include "AudioTools.h"
#include "SnapClient.h"
#include <SPI.h>

#include <Ethernet.h>

#define ETHERNET_MAC            "BA:E5:E3:B1:44:DD" // Ethernet MAC address (have to be unique between devices in the same network)

#define ETHERNET_RESET_PIN     21      // ESP32 pin where reset pin from W5500 is connected
#define ETHERNET_CS_PIN         5       // ESP32 pin where CS pin from W5500 is connected

#define ETH_PHY_ADDR 1
#define ETH_PHY_CS   5
#define ETH_PHY_IRQ  22

#define ETH_SPI_SCK  18
#define ETH_SPI_MISO 23
#define ETH_SPI_MOSI 19

NetworkClient ethernet;

WAVDecoder pcm;
I2SStream out;
SnapTimeSyncDynamic synch(172, 10); // optional configuration
SnapClient client(ethernet, out, pcm);

void ethernetWizReset(const uint8_t resetPin) {
    pinMode(resetPin, OUTPUT);
    digitalWrite(resetPin, HIGH);
    delay(250);
    digitalWrite(resetPin, LOW);
    delay(50);
    digitalWrite(resetPin, HIGH);
    delay(350);
}

void macCharArrayToBytes(const char* str, byte* bytes) {
    for (int i = 0; i < 6; i++) {
        bytes[i] = strtoul(str, NULL, 16);
        str = strchr(str, ':');
        if (str == NULL || *str == '\0') {
            break;
        }
        str++;
    }
}

void connectEthernet() {
    byte mac[6];
    macCharArrayToBytes(ETHERNET_MAC, mac);
    Ethernet.init(ETHERNET_CS_PIN);
    ethernetWizReset(ETHERNET_RESET_PIN);

    if (Ethernet.begin(mac) == 0) {
        Serial.println("Failed to configure Ethernet using DHCP");
    } else {
        Serial.print("Ethernet IP is: ");
        Serial.println(Ethernet.localIP());
    }
}

void setup() {
    Serial.begin(115200);

    SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
    connectEthernet();

    auto cfg = out.defaultConfig(TX_MODE);
    cfg.pin_bck = 21;
    cfg.pin_ws = 19;
    cfg.pin_data = 26;
    cfg.sample_rate = 44100; // Set the sample rate to 44.1 kHz
    cfg.bits_per_sample = 16;  // Set bits per sample to 16
    cfg.i2s_format = I2S_PHILIPS_FORMAT;  // Stereo channel, Philips format (I2S)
    cfg.use_apll = true;  // Use the audio PLL for low-jitter
    cfg.is_master = true;  // Set to master mode
    out.begin(cfg);

    client.begin(synch);
}

void loop() {
    client.doLoop();
    delay(10);  // Add a small delay to prevent rapid restarts and allow for debug logging
}

Core  1 register dump:
PC      : 0x400d7834  PS      : 0x00060630  A0      : 0x800dcb6c  A1      : 0x3ffb21a0  
A2      : 0x3ffc39e8  A3      : 0x00001770  A4      : 0xffffffff  A5      : 0x3ffc50e8  
A6      : 0x0000174c  A7      : 0x00000004  A8      : 0x800d7832  A9      : 0x3ffb2120  
A10     : 0x00000000  A11     : 0x00000044  A12     : 0x3f401d16  A13     : 0x00000117  
A14     : 0x3f401d28  A15     : 0x3f402ff4  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008b04d  LEND    : 0x4008b05d  LCOUNT  : 0xfffffffc  

Backtrace: 0x400d7831:0x3ffb21a0 0x400dcb69:0x3ffb2210 0x400dcc50:0x3ffb2230 0x400d8263:0x3ffb2250 0x400ea1d4:0x3ffb2270 0x40090772:0x3ffb2290

pschatzmann, 2024-07-17:
What do you mean there is not enough memory: You just forgot to define the partition scheme.
Why don't you analyse the back trace? This gives you the info where it is crashing...

parabyte, 2024-07-17:
sorry i figured out that out once you pointed out about the partition scheme being wrong and now i have lots of room!

im abit over my head as they say with this, 

hope this information helps 

this is for the 2nd code i pasted, as the code with the ETH.H library does not even achieve dhcp.

(gdb) info line *0x400dce5d
Line 187 of "/home/gatekeeper/Arduino/libraries/arduino-snapclient/src/api/SnapProcessor.h"
   starts at address 0x400dce5b <_ZN12snap_arduino13SnapProcessor15processLoopStepEv+3>
   and ends at 0x400dce60 <_ZN12snap_arduino13SnapProcessor15processLoopStepEv+8>.
(gdb) info line *0x400dcf44
Line 78 of "/home/gatekeeper/Arduino/libraries/arduino-snapclient/src/api/SnapProcessor.h"
   starts at address 0x400dcf44 <_ZN12snap_arduino13SnapProcessor6doLoopEv+20> and ends at 0x400dcf47 <_ZN12snap_arduino13SnapProcessor6doLoopEv+23>.
(gdb) info line *0x400d846d
Line 159 of "/home/gatekeeper/Arduino/libraries/arduino-snapclient/src/SnapClient.h" starts at address 0x400d8469 <_Z4loopv+9> and ends at 0x400d8470 <_Z4loopv+16>.
(gdb) info line *0x400eaa20
Line 74 of "/home/arduino/arduino/packages/esp32/hardware/esp32/3.0.2/cores/esp32/main.cpp" starts at address 0x400eaa20 <_Z8loopTaskPv+48>
   and ends at 0x400eaa23 <_Z8loopTaskPv+51>.
(gdb) info line *0x40090772
Line 162 of "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c"
   starts at address 0x4009076f <vPortTaskWrapper+3> and ends at 0x40090775 <vPortTaskWrapper+9>.

parabyte, 2024-07-17:
and i corrected the i2s pins in my local code here, i overlooked that and unsure why there were such absurd values, perhaps my dog or cat got on my keyboard!