pschatzmann/arduino-snapclient #26: ethernet issue

parabyte, 2024-07-17:
code is bootlooping,it does achieve dhcp as i checked my dhcp server. below code is the backtrace

#include "AudioTools.h"
#include "SnapClient.h"
#define ARDUINO_LOOP_STACK_SIZE (10 * 1024)
#include <Ethernet.h>
#include <SPI.h>

#define ETHERNET_MAC            "BA:E5:E3:B1:44:DD" // Ethernet MAC address (have to be unique between devices in the same network)
#define ETHERNET_IP             "192.168.10.15"       // IP address of RoomHub when on Ethernet connection

#define ETHERNET_RESET_PIN     21      // ESP32 pin where reset pin from W5500 is connected
#define ETHERNET_CS_PIN         5       // ESP32 pin where CS pin from W5500 is connected

#define ETH_PHY_ADDR 1
#define ETH_PHY_CS   5
#define ETH_PHY_IRQ  22

// SPI pins
#define ETH_SPI_SCK  18
#define ETH_SPI_MISO 23
#define ETH_SPI_MOSI 19

EthernetClient ethernet;

WAVDecoder pcm;
I2SStream out;
SnapTimeSyncDynamic synch(172, 10); // optional configuration
SnapClient client(ethernet, out, pcm);

void ethernetWizReset(const uint8_t resetPin) {
    pinMode(resetPin, OUTPUT);
    digitalWrite(resetPin, HIGH);
    delay(250);
    digitalWrite(resetPin, LOW);
    delay(50);
    digitalWrite(resetPin, HIGH);
    delay(350);
}

void macCharArrayToBytes(const char* str, byte* bytes) {
    for (int i = 0; i < 6; i++) {
        bytes[i] = strtoul(str, NULL, 16);
        str = strchr(str, ':');
        if (str == NULL || *str == '\0') {
            break;
        }
        str++;
    }
}

void connectEthernet() {
    byte* mac = new byte[6];
    macCharArrayToBytes(ETHERNET_MAC, mac);
    Ethernet.init(ETHERNET_CS_PIN);
    ethernetWizReset(ETHERNET_RESET_PIN);
  
    Serial.println("Starting ETHERNET connection...");
    Ethernet.begin(mac);

    Serial.print("Ethernet IP is: ");
    Serial.println(Ethernet.localIP());
}

void setup() {
    Serial.begin(115200);

    SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
    connectEthernet();

    auto cfg = out.defaultConfig(TX_MODE);
    cfg.pin_bck = 21;
    cfg.pin_ws = 19;
    cfg.pin_data = 26;
    cfg.sample_rate = 44100; // Set the sample rate to 44.1 kHz
    cfg.bits_per_sample = 16;  // Set bits per sample to 16
    cfg.i2s_format = I2S_PHILIPS_FORMAT;  // Stereo channel, Philips format (I2S)
    cfg.use_apll = true;  // Use the audio PLL for low-jitter
    cfg.is_master = true;  // Set to master mode
    out.begin(cfg);

    client.begin(synch);
}

void loop() {
    client.doLoop();
}

 this is backtrace

(gdb) info line *0x400dce5d
Line 187 of "/home/gatekeeper/Arduino/libraries/arduino-snapclient/src/api/SnapProcessor.h"
   starts at address 0x400dce5b <_ZN12snap_arduino13SnapProcessor15processLoopStepEv+3>
   and ends at 0x400dce60 <_ZN12snap_arduino13SnapProcessor15processLoopStepEv+8>.

(gdb) info line *0x400dcf44
Line 78 of "/home/gatekeeper/Arduino/libraries/arduino-snapclient/src/api/SnapProcessor.h"
   starts at address 0x400dcf44 <_ZN12snap_arduino13SnapProcessor6doLoopEv+20> and ends at 0x400dcf47 <_ZN12snap_arduino13SnapProcessor6doLoopEv+23>.

(gdb) info line *0x400d846d
Line 159 of "/home/gatekeeper/Arduino/libraries/arduino-snapclient/src/SnapClient.h" starts at address 0x400d8469 <_Z4loopv+9> and ends at 0x400d8470 <_Z4loopv+16>.

(gdb) info line *0x400eaa20
Line 74 of "/home/arduino/arduino/packages/esp32/hardware/esp32/3.0.2/cores/esp32/main.cpp" starts at address 0x400eaa20 <_Z8loopTaskPv+48>
   and ends at 0x400eaa23 <_Z8loopTaskPv+51>.

(gdb) info line *0x40090772
Line 162 of "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c"
   starts at address 0x4009076f <vPortTaskWrapper+3> and ends at 0x40090775 <vPortTaskWrapper+9>.

parabyte, 2024-07-18:
i corrected my pin error i made in the code and i still have the exact same error

pschatzmann, 2024-07-18:
It is hard to believe that it is stopping at Line 187  of SnapProcessor.h. If you double check this is just the method connectClient()!

I noticed that I am doing the connect in the loop processing, so I decided to do change the code so, that the initial connection to the server is already done in begin(), so that you can check if this was successful and you would need to stop if not.

Please look at the error checking logic of this [DRAFT Ethernet example](https://github.com/pschatzmann/arduino-snapclient/blob/main/examples/SnapClientEthernet/SnapClientEthernet.ino)

pschatzmann, 2024-07-18:
Did you install the updated library version ?

parabyte, 2024-07-18:
yes

parabyte, 2024-07-18:
yes

pschatzmann, 2024-07-18:
Did you also stop the processing when begin did not return true ?

parabyte, 2024-07-18:
i just went with your example you posted above as i figured you wrote it to debug, so i adapted it with the pins of my cheap yellow display pinout "esp32 device" im not getting a backtrace with this code annoyingly just a coredump, im unsure if that be of any use to you,

parabyte, 2024-07-18:
i do not think the ethernet.h library is compatible with the esp32?

pschatzmann, 2024-07-18:
Did you manage to have any Ethernet example provided by Arduino working ?

parabyte, 2024-07-18:
the only ethernet example code that functions on the esp32 seems to be the one provided by espressif,  i have had the w5500 variant functioning, will try ETH.h library again  shortly with your code see what happens

parabyte, 2024-07-18:
#include <SPI.h>
#include <ETH.h>
#include "AudioTools.h"
#include "SnapClient.h"

#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS   5
#define ETH_PHY_IRQ  22
#define ETH_PHY_RST  21

// SPI pins
#define ETH_SPI_SCK  18
#define ETH_SPI_MISO 23
#define ETH_SPI_MOSI 19

NetworkClient eth;
HexDumpOutput out;  // final output
CopyDecoder codec;
SnapClient client(eth, out, codec);
byte mac[] = {0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED};
IPAddress ip(192, 168, 1, 177);

void setup() {
  Serial.begin(115200);

  // The ESP32 supports a flexible definition of the SPI pins
  //SPI.begin(ETH_SCLK, ETH_MISO, ETH_MOSI, ETH_CS);

  SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);

  // start the Ethernet connection:
  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);

  client.setServerIP(IPAddress(192, 168, 1, 38));

  }  

  
void loop() { client.doLoop(); }

 info line *0x400d5bc1
Line 287 of "/home/gatekeeper/Arduino/libraries/arduino-snapclient/src/api/SnapProcessor.h"
   starts at address 0x400d5b8e <_ZN12snap_arduino13SnapProcessor13connectClientEv+18> and ends at 0x400d5bc2 <_ZN12snap_arduino13SnapProcessor13connectClientEv+70>.
(gdb) info line *0x400da7d9
Line 194 of "/home/gatekeeper/Arduino/libraries/arduino-snapclient/src/api/SnapProcessor.h"
   starts at address 0x400da7d7 <_ZN12snap_arduino13SnapProcessor15processLoopStepEv+3>
   and ends at 0x400da7df <_ZN12snap_arduino13SnapProcessor15processLoopStepEv+11>.
(gdb) info line *0x400da8c0
Line 85 of "/home/gatekeeper/Arduino/libraries/arduino-snapclient/src/api/SnapProcessor.h"
   starts at address 0x400da8c0 <_ZN12snap_arduino13SnapProcessor6doLoopEv+20> and ends at 0x400da8c3 <_ZN12snap_arduino13SnapProcessor6doLoopEv+23>.
(gdb) info line *0x400da8d1
Line 159 of "/home/gatekeeper/Arduino/libraries/arduino-snapclient/src/SnapClient.h" starts at address 0x400da8cb <_Z4loopv+3> and ends at 0x400da8d4 <_Z4loopv+12>.
(gdb) info line *0x400e8704
Line 74 of "/home/arduino/arduino/packages/esp32/hardware/esp32/3.0.2/cores/esp32/main.cpp" starts at address 0x400e8704 <_Z8loopTaskPv+48>
   and ends at 0x400e8707 <_Z8loopTaskPv+51>.
(gdb) info line *0x40090682
Line 162 of "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c"
   starts at address 0x4009067f <vPortTaskWrapper+3> and ends at 0x40090685 <vPortTaskWrapper+9>.

parabyte, 2024-07-18:
just to confirm things, the ETH.h does not get as far as getting dhcp from my router like ethernet.h did

pschatzmann, 2024-07-19:
When posting code, please add the missing code tags: This is very hard to read!
 
You forgot to call the following in setup:

```C++
  // start snap client
  if (!client.begin()) {
    Serial.print("Could not connect to snap server");
    while (true);
  }
```

You also seem to contradict yourself: You say that the Arduino ETH.h example was working!

parabyte, 2024-07-19:
yes sorry i am not very good at explaining things, basically if i configure the espressif ethernet  demo code that comes with there libs i can use there code and it pings google or something i forget, as it is that works, with your code it does nothing at all and produces the result above, with the ethernet.h stuff i cannot seem to get there demo's to function on my esp32, but if i use ethernet.h with your code, it grabs a ip address from my dhcp server, 

sorry for the confusion, will add what you suggested to the code, thanks for bearing with me.

pschatzmann, 2024-07-19:
So you need to make sure that you execute client.begin()  but only after you have connected! 
You can achieve this e.g by waiting for eth_connected to be come true as I mentioned above!

So the following might also work:
```C++
void setup() {
Serial.begin(115200);

SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);

// wait for ETH to become active
while(ETH.linkUp()){
   delay(100);
}

client.setServerIP(IPAddress(192, 168, 1, 38));
  // start snap client
  if (!client.begin()) {
    Serial.print("Could not connect to snap server");
    while (true);
  }

}
```

pschatzmann, 2024-07-19:
Please note that I made some corrections to the code above!

parabyte, 2024-07-19:
im very confident you have gotten everything working correctly and its just my own human stupidly! i must add i am not a coder at all. this code works without throwing any bootloops just not connecting to the snap server now. am getting dhcp can ping its ip now, just verified that. #include <SPI.h>

#include "AudioTools.h"
#include "SnapClient.h"
#include <ETH.h>

#define USE_TWO_ETH_PORTS 0

#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS   5
#define ETH_PHY_IRQ  22
#define ETH_PHY_RST  21

// SPI pins
#define ETH_SPI_SCK  18
#define ETH_SPI_MISO 23
#define ETH_SPI_MOSI 19

 NetworkClient ethernet;

WAVDecoder pcm;
I2SStream out;
SnapTimeSyncDynamic synch(172, 10); // optional configuration
SnapClient client(ethernet, out, pcm);

static bool eth_connected = false;

void connectEthernet() {

  //  ethernetWizReset(RESET);
  
   // Serial.println("Starting ETHERNET connection...");
  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);
        ETH.setHostname("esp32-eth0");

  // wait for link
  while(!ETH.linkUp()){
   delay(200);
  }

    Serial.print("Ethernet IP is: ");
    Serial.println(ethernet.localIP());
}

void setup() {
    //Serial.begin(115200);

  SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
    connectEthernet();

    auto cfg = out.defaultConfig(TX_MODE);
    cfg.pin_bck = 25;
    cfg.pin_ws = 27;
    cfg.pin_data = 26;
    cfg.sample_rate = 44100; // Set the sample rate to 44.1 kHz
    cfg.bits_per_sample = 16;  // Set bits per sample to 16
    cfg.i2s_format = I2S_PHILIPS_FORMAT;  // Stereo channel, Philips format (I2S)
    cfg.use_apll = true;  // Use the audio PLL for low-jitter
    cfg.is_master = true;  // Set to master mode
    out.begin(cfg);

   // client.begin(synch);
  client.setServerIP(IPAddress(192, 168, 10, 103));
      // start snap client
  if (!client.begin()) {
    Serial.print("Could not connect to snap server");
    while (true);
  }
}

void loop() {
    client.doLoop();
}
and output from serial
03:00:19.340 -> [   563][E][NetworkClient.cpp:636] localIP(): NetworkClient::localIP Not AF_INET or AF_INET6?
03:00:19.340 -> [   572][E][SnapClient.h:98] begin(): [SnapClient] WiFi not connected
03:06:17.244 -> ����������������������[  2662][E][NetworkClient.cpp:636] localIP(): NetworkClient::localIP Not AF_INET or AF_INET6?
03:06:19.282 -> [  2672][E][SnapClient.h:98] begin(): [SnapClient] WiFi not connected

parabyte, 2024-07-19:
just want to explain that i am using networkclass as ETH.h does not provide EthernetClient class

pschatzmann, 2024-07-19:
I committed a correction that will  login to Wifi only when a WiFiClient has been provided.
This should fix the "WiFi not connected" error

parabyte, 2024-07-19:
i even commented out the example wifi credentials in your library, with the same code getting same behaviour 3:59:33.710 -> [  4562][E][NetworkClient.cpp:636] localIP(): NetworkClient::localIP Not AF_INET or AF_INET6?
03:59:33.710 -> [  4572][E][SnapClient.h:119] begin(): [SnapClient] WiFi not connected

pschatzmann, 2024-07-19:
Did you install the new library version ?

parabyte, 2024-07-19:
yes i pulled it from git as soon as i saw the changes go live

pschatzmann, 2024-07-19:
Strange: I will need to check why this is not working. 

As a work-around I have also added the method setWiFi() so that you can call this in your sketch to deactivate the automatic login: client.setWiFi(false);

parabyte, 2024-07-19:
968][V][ETH.cpp:104] _onEthEvent(): eth0 Started
04:15:56.609 -> [   984][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 1 - ETH_START
04:15:56.641 -> [   984][V][esp32-hal-periman.c:174] perimanSetPinBusExtraType(): Successfully set extra_type ETH_IRQ for pin 22
04:15:56.641 -> [  1005][V][esp32-hal-periman.c:160] perimanSetPinBus(): Pin 21 successfully set to type ETHERNET_SPI (48) with bus 0x3ffc456c
04:15:56.641 -> [  1019][V][esp32-hal-periman.c:174] perimanSetPinBusExtraType(): Successfully set extra_type ETH_RST for pin 21
04:15:58.600 -> [  2969][V][ETH.cpp:95] _onEthEvent(): eth0 Connected
04:15:58.600 -> [  2974][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 3 - ETH_CONNECTED
04:15:58.600 -> [  2976][V][NetworkInterface.cpp:66] _onIpEvent(): eth0 Got New IP: 192.168.10.171 MASK: 255.255.255.0 GW: 192.168.10.254
04:15:58.632 -> [  2994][V][NetworkEvents.cpp:119] checkForEvent(): Network Event: 5 - ETH_GOT_IP
04:15:58.664 -> [  3031][E][NetworkClient.cpp:636] localIP(): NetworkClient::localIP Not AF_INET or AF_INET6?
04:15:58.664 -> [  3040][E][SnapClient.h:119] begin(): [SnapClient] WiFi not connected

pschatzmann, 2024-07-19:
I see: WiFiClient is just a typedef of NetworkClient: so this is the same thing.
The only way to fix this is to call client.setWiFi(false); before begin()

parabyte, 2024-07-19:
i can confirm that client.setWiFi(false);  works very nicely, Thank you for fixing this, :)

closed 2024-07-19 by parabyte