pschatzmann/arduino-snapclient #27: ethernet

parabyte, 2024-07-19:
I just wish to confirm that ethernet.h library i cannot get to work with the esp32, crashes your code in a non stop bootloop.

same goes for there example code they ship.

The ethernet library ETH.H that espressif ships works perfectly with your code.

parabyte, 2024-07-19:
this is what i am using for ethernet and its working perfectly, #include <SPI.h>

#include <SPI.h>
#include "AudioTools.h"
#include "SnapClient.h"
#include <ETH.h>

#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS   5
#define ETH_PHY_IRQ  22
#define ETH_PHY_RST  21

// SPI pins
#define ETH_SPI_SCK  18
#define ETH_SPI_MISO 23
#define ETH_SPI_MOSI 19

#  define CONFIG_SNAPCAST_BUFF_LEN 2048

NetworkClient ethernet;
WAVDecoder pcm;
I2SStream out;
SnapTimeSyncDynamic synch(172, 10); // optional configuration
SnapClient client(ethernet, out, pcm);

static bool eth_connected = false;

void connectEthernet() {

  //  ethernetWizReset(RESET);
  
   // Serial.println("Starting ETHERNET connection...");
  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);
        ETH.setHostname(CONFIG_SNAPCAST_CLIENT_NAME);

  // wait for link
  while(!ETH.linkUp()){
   delay(100);
  }

    Serial.print("Ethernet IP is: ");
    Serial.println(ethernet.localIP());
}

void setup() {
    //Serial.begin(115200);
client.setWiFi(false);
  SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
    connectEthernet();

    auto cfg = out.defaultConfig(TX_MODE);
    cfg.pin_bck = 27;
    cfg.pin_ws = 26;
    cfg.pin_data = 25;
    cfg.sample_rate = 44100; // Set the sample rate to 44.1 kHz
    cfg.bits_per_sample = 16;  // Set bits per sample to 16
    cfg.i2s_format = I2S_PHILIPS_FORMAT;  // Stereo channel, Philips format (I2S)
    cfg.use_apll = true;  // Use the audio PLL for low-jitter
    cfg.is_master = true;  // Set to master mode
    out.begin(cfg);

   // client.begin(synch);
  client.setServerIP(IPAddress(192, 168, 10, 103));
      // start snap client
  if (!client.begin(synch)) {
    Serial.print("Could not connect to snap server");
    while (true);
  }
}

void loop() {
    client.doLoop();
}

parabyte, 2024-07-19:
This is what does work perfectly with your library 

#include <SPI.h>
#include "AudioTools.h"
#include "SnapClient.h"
#include <ETH.h>

#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS   5
#define ETH_PHY_IRQ  22
#define ETH_PHY_RST  21

// SPI pins
#define ETH_SPI_SCK  18
#define ETH_SPI_MISO 23
#define ETH_SPI_MOSI 19

#  define CONFIG_SNAPCAST_BUFF_LEN 2048
NetworkClient ethernet;
WAVDecoder pcm;
I2SStream out;
SnapTimeSyncDynamic synch(172, 10); // optional configuration
SnapClient client(ethernet, out, pcm);

static bool eth_connected = false;

void connectEthernet() {

  //  ethernetWizReset(RESET);
  
   // Serial.println("Starting ETHERNET connection...");
  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);
        ETH.setHostname(CONFIG_SNAPCAST_CLIENT_NAME);

  // wait for link
  while(!ETH.linkUp()){
   delay(100);
  }

    Serial.print("Ethernet IP is: ");
    Serial.println(ethernet.localIP());
}

void setup() {
    //Serial.begin(115200);
client.setWiFi(false);
  SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
    connectEthernet();

    auto cfg = out.defaultConfig(TX_MODE);
    cfg.pin_bck = 27;
    cfg.pin_ws = 26;
    cfg.pin_data = 25;
    cfg.sample_rate = 44100; // Set the sample rate to 44.1 kHz
    cfg.bits_per_sample = 16;  // Set bits per sample to 16
    cfg.i2s_format = I2S_PHILIPS_FORMAT;  // Stereo channel, Philips format (I2S)
    cfg.use_apll = true;  // Use the audio PLL for low-jitter
    cfg.is_master = true;  // Set to master mode
    out.begin(cfg);

   // client.begin(synch);
  client.setServerIP(IPAddress(192, 168, 10, 103));
      // start snap client
  if (!client.begin(synch)) {
    Serial.print("Could not connect to snap server");
    while (true);
  }
}

void loop() {
    client.doLoop();
}

parabyte, 2024-07-20:
this is working code using your library.

#include <SPI.h>
#include "AudioTools.h"
#include "SnapClient.h"
#include <ETH.h>

#define ETH_PHY_TYPE ETH_PHY_W5500
#define ETH_PHY_ADDR 1
#define ETH_PHY_CS   5
#define ETH_PHY_IRQ  22
#define ETH_PHY_RST  21

// SPI pins
#define ETH_SPI_SCK  18
#define ETH_SPI_MISO 23
#define ETH_SPI_MOSI 19

#  define CONFIG_SNAPCAST_BUFF_LEN 2048

NetworkClient ethernet;
WAVDecoder pcm;
I2SStream out;
SnapTimeSyncDynamic synch(172, 10); // optional configuration
SnapClient client(ethernet, out, pcm);

static bool eth_connected = false;

void connectEthernet() {

  
   // Serial.println("Starting ETHERNET connection...");
  ETH.begin(ETH_PHY_TYPE, ETH_PHY_ADDR, ETH_PHY_CS, ETH_PHY_IRQ, ETH_PHY_RST, SPI);
        ETH.setHostname(CONFIG_SNAPCAST_CLIENT_NAME);

  // wait for link
  while(!ETH.linkUp()){
   delay(100);
  }

    Serial.print("Ethernet IP is: ");
    Serial.println(ethernet.localIP());
}

void setup() {
    //Serial.begin(115200);
client.setWiFi(false);
  SPI.begin(ETH_SPI_SCK, ETH_SPI_MISO, ETH_SPI_MOSI);
    connectEthernet();

    auto cfg = out.defaultConfig(TX_MODE);
    cfg.pin_bck = 27;
    cfg.pin_ws = 26;
    cfg.pin_data = 25;
    cfg.sample_rate = 44100; // Set the sample rate to 44.1 kHz
    cfg.bits_per_sample = 16;  // Set bits per sample to 16
    cfg.i2s_format = I2S_PHILIPS_FORMAT;  // Stereo channel, Philips format (I2S)
    cfg.use_apll = true;  // Use the audio PLL for low-jitter
    cfg.is_master = true;  // Set to master mode
    out.begin(cfg);

   // client.begin(synch);
  client.setServerIP(IPAddress(192, 168, 10, 103));
      // start snap client
  if (!client.begin(synch)) {
    Serial.print("Could not connect to snap server");
    while (true);
  }
}

void loop() {
    client.doLoop();
}

parabyte, 2024-07-20:
sorry about the formatting #  define CONFIG_SNAPCAST_BUFF_LEN 2048
 is fine before i cut and paste the code here

pschatzmann, 2024-07-20:
Here are the instructions for pasting code:
https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks

What ESP32 Arduino core version are you using ?