pschatzmann/arduino-snapclient #28: clicking audio with ethernet

parabyte, 2024-07-19:
this is a issue i never had with wifi, i am having like a slight ticking sound that is regular. my brother suggested it could be a buffer issue he heard the issue too. I am really unsure how to diagnose or provide anything about this fault as there is  no error reports or problems to be reported from logs

parabyte, 2024-07-19:
okay i figured out the issue myself, was a lack of buffer cause were using ethernet now. 
be really lovely if you could explain why setting #  define CONFIG_SNAPCAST_BUFF_LEN 2048 sorts out my issue, just so i understand, many thanks

parabyte, 2024-07-20:
iv tried  # define CONFIG_SNAPCAST_BUFF_LEN 2048  and it does  not seem to help the issue, the issue seems to be some sort of memmory leak, if i power cycle my device the clicking does not accur initially, takes approx 2 minutes or so to happen, again this issue was not present when i was using wifi, so the issue is not on my desktop

parabyte, 2024-07-20:
iv tried # define CONFIG_SNAPCAST_BUFF_LEN 2048 and it does not seem to help the issue, the issue seems to be some sort of memmory leak, if i power cycle my device the clicking does not accur initially, takes approx 8 minutes or so to happen, again this issue was not present when i was using wifi, so the issue is not on my desktop

pschatzmann, 2024-07-20:
My guess would be that the Ethernet thruput is not fast enough for your selected settings.
- You can try to increase the SPI frequency or
- optimize the settings in /etc/snapserver.conf