pschatzmann/arduino-snapclient #39: SnapProcessor Socket connect to server failed when using Ethernet

Wardstein, 2024-10-23:
Hi, 
thanks first for the awesome lib. Just trying to get started with the snapclient. With WiFi on the ESP32 it works, but I cannot get the audio 100% glitch and crack free. Thats why I now want to move to Ethernet.
I have a W5500 board, connected via SPI. That works fine, I can ping my board.

First thing I noticed when moving to Ethernet instead of wifi, is that mDNS does not work (prints an error on the serial at startup). So I disabled it (via `#define CONFIG_SNAPCLIENT_USE_MDNS    false`) after reading that it only supports WiFi in the current implementation?

After calling `client.begin()` the SnapClient cannot connect to my snapcast server. All I get on the Serial output is
```
[   628][E][SnapProcessor.h:296] connectClient(): [SnapProcessor] Socket connect to 192.168.10.18:1704 failed (errno = 0)
```

Do you have any hints? I looked at the other Ethernet related issues here, but cannot find any significant difference in the code.

This is my code:
```
#include <Arduino.h>
#include <SPI.h>
#include <Ethernet.h>
#include "config.h"         // Must be above "SnapClient.h" include, as here are some defines, which are otherwise defaulted in "SnapClient.h/SnapConfig.h"
#include "AudioTools.h"
#include "SnapClient.h"
#include "AudioTools/AudioCodecs/CodecOpus.h"

EthernetClient eth;
byte mac[] = {0xC0, 0x49, 0xEF, 0xCE, 0x14, 0xD4};

AudioInfo infoOut(48000, 2, 16); 
OpusAudioDecoder codec;
I2SStream out;
SnapTimeSyncDynamic synch(172, 10); 
SnapClient client(eth, out, codec);

IPAddress ip(192, 168, 10, 99);
IPAddress gateway(192, 168, 10, 1);
IPAddress dns(192, 168, 10, 1);

void setup()
{
    Serial.begin(115200);

    Serial.println("Start SPI ...");
    SPI.begin(18, 19, 23, 5);

    // start the Ethernet connection:
    Serial.println("Start Ethernet ...");
    Ethernet.init(5);
    // Ethernet.begin(mac);
    Ethernet.begin(mac, ip, dns, gateway);

    // Check for Ethernet hardware present
    if (Ethernet.hardwareStatus() == EthernetNoHardware)
    {
        Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
        while (true);
    }
    Serial.print("Hardware type found: ");
    Serial.println(Ethernet.hardwareStatus());

    // Check link status
    if (Ethernet.linkStatus() == LinkOFF)
    {
        Serial.println("Ethernet cable is not connected.");
    }

    // wait for link
    Serial.println("waiting for link...");
    while(Ethernet.linkStatus() != LinkON)
    {
        delay(1000);
    }
    Serial.println("Ethernet link Up");

    Serial.print("local IP: ");
    Serial.println(Ethernet.localIP());

    Serial.println("starting I2S...");
    // setup I2S to define custom pins
    auto cfg = out.defaultConfig();
    cfg.copyFrom(infoOut);
    cfg.pin_data = 19;
    cfg.pin_bck = 18;
    cfg.pin_ws = 5;
    // cfg.pin_mck = 0;
    cfg.i2s_format = I2S_STD_FORMAT;
    cfg.buffer_size = 1024;
    cfg.buffer_count = 10;
    out.begin(cfg);

    Serial.println("starting SnapClient...");
    client.setWiFi(false);
    client.setServerIP(IPAddress(192, 168, 10, 18));
    client.begin(synch);

    Serial.println("started...");
}

void loop()
{
    client.doLoop();
}
```

pschatzmann, 2024-10-23:
My expectation would be that this should just work.

Double check the IP address of your snap server and make sure that there is no firewall blocking the server port (1704)

Wardstein, 2024-10-23:
Update: I traced it now down to the `readData()` function in `SnapProcessor.h`. It gets stuck in the loop
```
while (p_client->available() < base_message.size)
```

I guess this line "waits" for a complete message. According to the logs above and more tests, the `readBaseMessage()` function for example reads the header and the `size` field says 463 bytes. But a test output with `p_client->available()` in the the readData() functions returns only 34. No questions it waits here (for more I guess), but this never happens/arrives.

In this case my end question of the last comment is wrong, the server does not stop. But now what could this be, that there are no additional data received? Or the ESP32 is not receiving them.

Wardstein, 2024-10-24:
I got at least the network part working by now. Your hint with the incremental read was good. I rewrote the `SnapProcessor.h readData()` like this:
```
bool readData() {
    ESP_LOGD(TAG, "base_message.size: %d", base_message.size);
    if (base_message.size > send_receive_buffer.size()) {
      send_receive_buffer.resize(base_message.size);
    }
    start = &send_receive_buffer[0];
    int index = 0;
    do
    {
      int readLength = base_message.size;
      int available = p_client->available();
      if(available < readLength)
      {
        readLength = available;
      }
      if(available > readLength)
      {
        readLength = base_message.size - index;
      }
      readLength = p_client->readBytes(&(send_receive_buffer[index]), readLength);
      index += readLength;
    } while (index < base_message.size);
    size = base_message.size;

    return true;
  }
```

Now I get a stream of data, but this breaks up at some point. I checked this with tcpdump on my snapcast VM. The stream runs for under 10 seconds, then it breaks up (seems ACK packets from the ESP32 are missing then).

I poked a little bit around and it seems when I do not write the audio data into the output stream, everything runs as expected. But I do not understand why. Here is a list what I tried:

- Server settings with `chunk_ms` from 60 to 10 and `buffer` from 1000 to 10 do not solve it
- No matter which output I use (I2SStream, MeasuringStream) it stops at some point
- SnapProcessor.h processMessageLoop() `return true` after readData() call: works "forever" (left it running > 15min); ~100Hz main loop
- SnapProcessor.h processMessageWireChunk() `return true` before `switch(codec_from_server)`: works forever
- SnapProcessor.h wireChunk() `return true` after writeAudioInfo() call: works
- SnapProcessor.h writeAudio() `return size` instead `p_snap_output->write(): works
- SnapOutput.h write() `return size` before `if (!synchronizePlayback())`: works
- SnapOutput.h write() `return size` before `audioWrite()`: NOPE!
- SnapOutput.h audioWrite() `return size` before `decoder_stream.write()`: NOPE!

I looked at the other issues, especially #7, #12 and #14. But could not find a real solution. Tried also `SnapTimeSyncDynamic` and `SnapTimeSyncFixed`, but that does not change anything. I also varied the factor for `SnapTimeSyncFixed`, but this had no effect.
I enabled and disabled `CONFIG_SNAPCLIENT_SNTP_ENABLE` but this did not change anything.

Do you have any ideas why the stream breaks up, when I use a real audio stream output? Maybe a buffer over-/underrun? I see no error output on the serial.

pschatzmann, 2024-10-24:
Try it with RTOS processor: there you can play around with the buffer size.
Check the Ethernet library if there are some timeout values to be changed...

Wardstein, 2024-10-25:
Hmm, unfortunately the SnapProcessorRTOS does not change the behaviour. I tried it with 20kB at the output and also 20kB for the SnapProcessorRTOS.
Does the SnapProcessorRTOS only work with FreeRTOS? Or can I use it also without? Saw the example file where also no RTOS is used, so I guess I do not need to use it?
And do I need a board with PSRAM? I am currently working with a ESP32-WROOM-32.

pschatzmann, 2024-10-25:
No need for PSRAM, if you keep the buffer small enough...

Wardstein, 2024-10-25:
One thing I noticed during debugging with the RTOS processor, that I always get this output line:
`[ 14255][I][SnapProcessorRTOS.h:76] writeAudio(): [SnapProcessorRTOS] size: 531 / buffer 0`
which means `buffer.available()` is always 0. Also if the stream runs for a little longer and a few kB should be written to the buffer. 

And therefore as `buffer.available()` is always 0, the task_copy() is never started, so no output either. Tried to dig into the `buffer.writeArray` but this then ends in the espessif rtos lib.

I also tried a bit without the RTOS processor and found, that if I do not write into the opus_decode function in `CodecOpus.h write()`, then it also runs forever. So it seems it would need a buffer before the decoder?

Do you use any ESP32 board with ethernet? Can you recommend one where it works? This here seems like a dead end to me with my ESP32 and a W5500 ethernet chip.

pschatzmann, 2024-10-25:
So far I was only testing with WiFi and never had the time to look into any alternatives...
Maybe you give Wifi with RTOS a try...

Wardstein, 2024-11-19:
Could not get it to work glitch & noise free with WiFi either. 
Sadly due to other priorities I currently I cannot spend more time on this. But it is still a cool project I want to try in the future again. For now I will stick to my RasPi Zero W. Although it needs a lot more power :/

Thanks for all your you help and suggestions.

closed 2024-11-19 by Wardstein